awesomeshield/Awesome-Shield-PCB #2: Add second row of headers right next to the existing ones, with an offset of 100mil

aShure, 2015-05-11:


cwjpalmer, 2015-05-12:
What's the purpose of the headers? Should they be 'straight' aligned, or locking?

aShure, 2015-05-13:
I suggest straight ones, they can then be used to directly solder onto the shield for prototyping OR to populate female headers to accept another shield, while shipping out the product with sturdy and presoldered male headers. Male headers are cheaper and more robust than long stack male-female headers.

cwjpalmer, 2015-05-19:
We're actually leaning towards shipping with male headers, and not including a second row of headers. The rationale is:
- male headers make the components more physically accessible
- the vast majority of our users won't want to solder anything
- if people want to connect up additional components, they can always attach jumpers to the pins on the grove ports
- we want to keep the design simple, and avoid confusing people with stuff they don't need

closed 2015-05-19 by cwjpalmer