awesomeshield/Awesome-Shield-PCB #3: Run DRC and correct spacing,ie via next to R11

aShure, 2015-05-11:


cwjpalmer, 2015-05-13:
@aShure one DRC error I'm getting a lot is Drill size, one some vias I put into the ground pours to connect them to each other (last time I made a PCB I read that it's good practice to put vias into the ground planes). I used the default drill size for vias (outer diameter=1.0064, drill=0.6).

Questions:
1) should I include vias connecting the ground pours?
2) what size is appropriate?

aShure, 2015-05-13:
Check your net classes, it could be that the power class needs bigger values (which is wrong in this application, but might be the default)

1) Yes
2) same diameter as the holes that are used for the THT headers OR a via diameter that is the maximum which is tented by common PCB manufacturers. Tented via = buried and sealed under solder mask, hole partly or fully bridged by solder mask print.

cwjpalmer, 2015-05-19:
Okay, the spacing issues and other DRC issues are fixed, except for hole sizes on vias.

I'm still looking into tented via sizes, and then I'll update via sizes and make sure the DRC settings correspond. After that, the DRC should be happy :-)

cwjpalmer, 2015-05-20:
I heard back from one potential manufacturer (Seeed) and their max tented hole size is D=0.6mm. Coincidentally, that's what the hole size is currently.

@aShure Could you let me know what German manufacturer you're thinking about using for the first run of 3-4 test PCBS? I'll contact them to see what their max tented via size is. If it's smaller than D=0.6mm, I'll adjust the layout accordingly. If it's D>=0.6mm, I'll keep the vias at D=0.6mm.

Or, if you happen to know this number off hand, that would be cool, too :-)

We're getting close!

aShure, 2015-05-20:
Okay sounds good. 0.6mm sounds big for finished hole size, so I guess it's drill diameter.. I know at least http://www.multi-circuit-boards.eu/ don't guarantee tenting over 300µm finished hole size, so that's after plating.

I suggest https://www.haka-lp.de/ for the first run, please download the DRC file from their homepage and see if everything is okay. 
I think Haka have 2mm routing for free but they charge for anything smaller.

cwjpalmer, 2015-05-21:
@aShure Great!

My German is a little rough, so I want to confirm which of haka's DRU files I should use. I'm thinking "zwilling" because leo translates it as "twin". Is that right? Here are their options:

[image]

aShure, 2015-05-21:
haka_35_l2 and zwilling.

35µm copper, 2 layers and the budget version "zwilling" where you get a
pair of your prototype for a cheaper price if it can be processed in a
stack (hence the bigger diameter and worse tolerances, boards are
stacked and then drilled and routed - the more boards in the stack, the
lower the accuracy)

cwjpalmer, 2015-05-21:
Marvellous. Now the file nomenclature makes sense to me. Thanks for clarifying. And that's good to know about the 

I'll run the two DRC sets you mentioned, and make adjustments.

cwjpalmer, 2015-05-21:
**DRC results for haka's haka_35_l2 and zwilling**

all three branches ( `master`, `CPver2`, `CPvar6` ) had two sets of issues across both DRCs:
- solderable jumpers give 9 overlap errors on layer 1
- R9,R10 square arrangement gave 6 keepout errors, as already discussed and approved in issue 11

closed 2015-05-25 by cwjpalmer

aShure, 2015-05-28:
The DRC errors are all fine. And yes, you are correct that normal DRC settings are for untented vias. I will inquire about the tented via size.

aShure, 2015-05-28:
Can't reach Haka right now. 
Set the tented via size to e.g. 0.6mm - thus the headers won't be tented, but all the vias. If the vias can't be tented in production, they will be altered automatically.

cwjpalmer, 2015-05-28:
Thanks, Alex!