awesomeshield/Awesome-Shield-PCB #4: Check clearance for USB and power connector on Arduino motherboard, see if standard (short) male headers are suitable.

aShure, 2015-05-13:
Maybe clear route out the areas where the shield interferes with the motherboard. If possible in a pattern that gives no loss and interlocks two shields head to head, only separated with a 1mm routed path.

cwjpalmer, 2015-05-19:
My approach on this was to make the board a little smaller. Now it shouldn't overlap with the USB or power housings on an Arduino Uno. The new board layout should also work for interlocking head to head.

@aShure one question about interlocking: Right now, the board outline assumes that the centre of the router passes over the centre of the line in eagle on the board outline layer. This puts the angled section of the board outline exactly in the middle of the board, in the Y axis (as in "A" in the diagram). But I think that given the 1mm routed path, this angled section of the board outline should actually be offset by 0.5mm to correctly align 2 boards in the x-direction. Ie. it should be 0.5mm lower on the left hand board orientation, and 0.5mm higher on the right hand side board orientation, as in "B" in the diagram.

[image]

Does B make more sense to you, too? Are there other factors that I should account for?

aShure, 2015-05-19:
It's correct that the cnc router will ride along the middle of the drawn path as the paths middle is stored numerically as a 2d vector. It's important to choose the right size line width (1mm) for the line because that is passed to the machine as the end mill size. 

To me it seems that there is no offset needed if you choose the correct line width. You can verify the design if you a 180 degree rotation around the middle of the slope - the whole board outline has got to be point symmetric around that.

cwjpalmer, 2015-05-19:
Fantastic! So it's scenario A, with the correct line width (1mm). I'll verify by rotation as you've suggested.

cwjpalmer, 2015-05-19:
I checked branch `master`, `CPver2` and `CPvar6` by rotation. They all seem good.

cwjpalmer, 2015-05-19:
@aShure when I changed the route width to 1mm, the DRC had issues with a whole bunch of spacing around the edge of the board. Many things (eg. traces) are easy to fix, but it also had problems with half the pins on the locking headers. Do you recommend that I:

A) only change the interlocking edges between boards to 1mm router width, and leave the the other sides as they were initially
B) change to non-locking headers – it seems like this would fix the problem, but I hope there's a better solution, because the locking headers will make such a big difference during assembly
C) do something else?

Here's a screen shot of A, so you can see what I mean:
[image]

aShure, 2015-05-19:
The DRC rules might be a tad too conservative, however, after your decision to ditch the stacking headers, it makes sense to use non locking footprints for the male headers. It's a joy to solder them, if they are plugged in another shield or mainboard. The Awesomeshield then just rests on top and is perfectly flat with no strain. So my vote would be B) :+1: 

I suggest using the outline layer either on the middle of the path with 1mm width or offset to the inside about 0.5mm and then width a hairline width. In any case, use the *milling* layer and describe the contours there, stick to 1mm width, anything below 1mm as an outline is often just rejected. (some require even 2mm as it is easier to machine and more boards can be stacked!). Don't forget to add tabs, ie interruptions on the path, so that the board won't fall out and crash on the machine, but stay in place in the panel only to be separated after soldering. I suggest 0.5mm tabs at the top left, bottom left, mid right. You have to think about a layout that daisy chains and supports adjacent boards in the panel. Should I draft something to explain it further?

cwjpalmer, 2015-05-20:
Okay, Awesome! I implemented option B in the branches containing our three working versions.

Thanks for the info about the milling layer. It was completely empty. I'll fix that.

cwjpalmer, 2015-05-20:
Re: tab locations, thinking about the board being fit together in pairs, are you thinking tabs at these locations (tabs indicated by notches perpendicular to the outline)?

[image]

aShure, 2015-05-20:
That is an excellent variant, exactly what I had in mind! If you leave a little space towards the corners (just like you sketched), it's easier to get a clean corner after breaking them apart. Pay special attention that you don't deviate in the X axis with the vertical tabs, as they need to be daisy chained to the adjacent boards on the panel.

cwjpalmer, 2015-05-20:
okay great! so it'll look more like this:

[image]

aShure, 2015-05-20:
Looks perfect :+1:

cwjpalmer, 2015-05-20:
okay, routing is done, along with tabs.

closed 2015-05-20 by cwjpalmer