awesomeshield/Awesome-Shield-PCB #10: A3 is connected to GND instead of potentiometer

aShure, 2015-05-19:
check middle pin of potentiometer, it's NC right now.

cwjpalmer, 2015-05-20:
good catch! I connected A3 to the middle pin of the pot on `master` , `CPver2` , `CPvar6`

@cwjpalmer: remember to update this on `issue11` if merging that branch into main

cwjpalmer, 2015-05-20:
@aShure do you think I should connect the large mounting holes to GND, or leave them as NC?

aShure, 2015-05-20:
Connecting to GND is fine, use thermals to make it easy to solder - lots of thermal mass.

cwjpalmer, 2015-05-20:
@aShure I'm having a bit of a tough time getting eagle to connect those pads to the GND net. Is there any reason not to leave them NC (as they are now, and as the part in the Sparkfun library is configured)? If there's nothing bad about leaving them as is, I'll do that.

aShure, 2015-05-20:
No problem just leave them like that. FYI for future endeavors: If you
can't manage to connect the pin in the library part, you can always
route a wire with the GND net (use the name command) to a pin manually
as a hack and ignore the DRC error for that pin then.

cwjpalmer, 2015-05-21:
Great! I'll leave it, and that's a really good trick to know. Thanks for sharing the tip :-)

closed 2015-05-21 by cwjpalmer