awesomeshield/Awesome-Shield-PCB #11: rearrange R3,4 and R9,10 so that one single part can be used too

aShure, 2015-05-19:
Two resistors can be bridged if they are placed in a way that all of their four pads are in a regular square pattern. That way one can either use two in series or just one part, with half the resistance.

cwjpalmer, 2015-05-20:
I think I understand. I'll give it a try, and you can let me know if I did it correctly.

cwjpalmer, 2015-05-20:
Okay, I gave this a try in the branch `issue11` https://github.com/cwjpalmer/Awesome-Shield-PCB-0.5/tree/issue11

@aShure Is this what you had in mind? If so, let me know and I'll merge this back into master, and make similar changes in the other two working variants.

Some things that I noticed that might be problematic:
1. the spacing of the 0603 resistors (R9,10) now violates the DRC (in order to put the pads in the square arrangement)
2. the pads for the resistors aren't square. So although they're arranged in a square, if you tried to solder a single resistor between the two sets of pads (to avoid using 2 in series) I think that part of the resistor would land on solder mask. You can see what I mean by copying a resistor, rotating it 90 degrees, and holding over the two pads, where you would solder it.

aShure, 2015-05-20:
I am sure that the 1206 resistors will fit either way even if the pad is designed for reflow and not square. I guess you're onto something with the 0603 pads being not square. Does the DRC complain about the tRestrict layer with the 0603 parts or is it something else?

cwjpalmer, 2015-05-20:
That's great to know about the 1206 resistors. About the 0603 ones: the DRC complains about the tKeepout layer (39). Here's a screenshot:

[image]

aShure, 2015-05-20:
> keepout drc resistors

That's enough space for my taste, I did closer than that with Avoccado°.
The 0603 wouldn't be assembled manually anyways. One might push R9 away
with the tweezers while setting R10 with manual placement.

cwjpalmer, 2015-05-20:
Fantastic! So then it seems like I should incorporate what you see above (the resistor layout changes made in branch `issue11` into the 3 working variants. Does that makes sense to you? Currently, the square resistor layout is only present on branch `issue11`.

aShure, 2015-05-20:
Sure.
When I had a look at the branches it was not clear to me what the
differences are. Mind you I only took a quick glance at the
repositories, the branch names or READMEs where not explanatory to me. I
think you might be doing redundant work if you alter all branches
separately and might be faster off with sticking to one or two.

cwjpalmer, 2015-05-20:
Ya, I don't think I'm in a very efficient workflow. The three branches (master, CPver2, CPvar6) are three different layouts so that we can try different possibilities for UX with components in different places. I should have done one version, got you to check it, fixed all this stuff we're working through now, and then created the variants, I think. Now I know for next time :-)

cwjpalmer, 2015-05-20:
okay, all three variants now have the right resistor layouts for the R3,4,9,10

closed 2015-05-20 by cwjpalmer