awesomeshield/Awesome-Shield-PCB #13: Add fiducials

aShure, 2015-05-20:


aShure, 2015-05-22:
nudge the fiducials onto the grid, micro fiducials should align with the components' mid points. This is unresolved as of now, as some components are not placed on the grid.

aShure, 2015-05-22:
correct placement shall be finished by @cwjpalmer :-)

cwjpalmer, 2015-05-23:
will do!

cwjpalmer, 2015-05-23:
@aShure just to clarify, can you please confirm or correct this stuff? :
1. Macro fiducials should be on mm grid, near where you put them
2. Micro fiducials should be on mm grid, at midpoints of components
2b. Question: do only SMD / machine placed components need micro fiducials?
3. Should fiducials not overlap with silkscreen, or does that matter?

aShure, 2015-05-23:
1. Yes 
2. Local component fiducials are not always necessary, but as there's so much space I placed them at the odd shaped components. Midpoint or one edge or three outer edges are common practice. 
None of the components actually need them on this board. 
Yes, metric grid please when possible - most probably that will not be the case for the mid point. 

Fiducials should be free of silkscreen and solder mask, just bare copper with an annulus of nothing around them, the more space the better.

cwjpalmer, 2015-05-23:
Thanks! That was really clear and really helpful :-)

cwjpalmer, 2015-05-23:
Okay, I think it's done correctly in commit `7b772e0cbf31359f01a4ac244fdb73c1b373331e`

closed 2015-05-23 by cwjpalmer