awesomeshield/Awesome-Shield-PCB #19: height should be 53 or 52, makes it easier to panelize

aShure, 2015-05-24:


aShure, 2015-05-24:
width should be reduced to 49, there's redundant space on both sides. I would take it away from the right side.
Just an example:
with a board spacing of 1mm and a width of 100mm per panel, the board better be less than 49.5mm. For a 2mm router bit, it would need to be no bigger than 49mm. If scaling up to more than just two boards in a row, each iteration should occupy no more than a 50mm including the milling bit width.

There might be space for 48mm..

cwjpalmer, 2015-05-25:
height is at 53mm, board outline is on mm grid as of commit `d2fe7ea03bfe87debf3c3968f5a6bcddb1743020`

cwjpalmer, 2015-05-25:
adjusted board width to 49mm in commit `23dbb87129958a432bd4e2d01da165933cc024d0`

@aShure there's not much substrate left between the top row of header holes (ca. 1.7mm between the outside holes and the edge of the board). Do you think that's okay? Do you think we can change the width to 48mm and still maintain strength? This would mean ca. 1.5mm from holes to the edge of the board.

aShure, 2015-05-28:
Yes that is enough. Scroll down at https://github.com/avoccado/avoccado-pcb-avr to see various examples of just <1.5mm hole to outline spacing

closed 2015-05-28 by aShure

cwjpalmer, 2015-05-28:
Awesome! Thanks! Also, those are some gorgeous layout! I think I'm only beginning to appreciate the beauty of the engineering, but they're also visually stunning images.