awesomeshield/Awesome-Shield-PCB #22: 75 Ohms resistors as current limiters are too low

aShure, 2015-05-27:
worst case for red: 
(5V-1.8V) / 2 x 75 Ohms = 0.022 A

That's above the spec of the LED and the portpins, if the shield should be compatible with the newest Arduinos. The Arudino Due can only use certain portpins with more than 3mA, the new Arduino Zero runs on 3.3V and has got a lower current capability than the Uno.

"DC Current per I/O Pin	7 mA" (http://www.arduino.cc/en/Main/ArduinoBoardZero)

Some LEDs are run at less than 1mA, like the status LEDs on the normal Arduino boards. There's no need to use 20mA, 7mA will be bright.

I guess there need to be different resistor values if you want to have the same brightness for 5V and 3.3V motherboards.

cwjpalmer, 2015-05-27:
Good catch!

I redid the calculations for the 5V @ 7mA, and got 330 Ohm resistors. Here are my worst case calculations:
red:
(5V-1.8V) / 2 x 330 Ohms = 4.8mA
green and blue:
(5V-2.8V) / 330 Ohms = 6.7mA

And at 3V3:
red:
(3.3V-1.8V) / 2 x 330 Ohms = 2.3mA
green and blue:
(3.3V-2.8V) / 330 Ohms = 1.5mA

Which I guess will be dimmer, but still fine, as you pointed out that  < 1mA would be okay.

I've updated the resistor values in eagle in commits `master – b5265fd21d6b22c6ef18f395876f5d5a0203b2bf ` and `rightSideUI – a9772cc1788059bfb65202da16813fd8abb637ed `

aShure, 2015-05-28:
You can decrease the resistance a bit, as 3.3V and 7mA is for the new Arduinos, all the old ones run at 5V and can supply something like 20-40mA. So I would spec it at 7mA at 3.3V and 20mA at 5V, whichever gives the highest resistance.

cwjpalmer, 2015-05-28:
Brilliant!

cwjpalmer, 2015-05-28:
Okay, resistor specs are updated!

I arrived at 130 +/- 5% Ohm resistors. This means lowest possible resistors should be 123.5 Ohm, and:

3V3, 7mA red:
I = (3.3-1.8)V / (2 x 123.5Ohm) = 6.1mA
*6.1mA < 7mA*

3V3, 7mA green, blue:
I = (3.3-2.8)V / (123.5Ohm) = 4.0mA
4.0mA < 7mA

5V, 20mA red:
I = (5-1.8)V / (2 x 123.5Ohm) = 13mA
13mA  < 20mA

5V, 20mA green, blue:
I = (5-2.8)V / (123.5Ohm) = 18mA
*18mA < 20mA*

These values are updated in commits:
master – 91ba2e49fd83367d4ef821892daf04396256fdae
rightSideUI – ff7fc9ccc651946b41404483b5a3b1b7c972cac0

closed 2015-05-28 by cwjpalmer