awesomeshield/Awesome-Shield-PCB #27: Validate that the buzzer is compatible for the new Arduino models

aShure, 2015-05-28:
Due to the current and voltage changes at the portpin levels of the new Arduinos, a current limiting resistor might be needed in a future proof design.

cwjpalmer, 2015-05-28:
As far as I can tell, buzzers have a coil resistance. To find this, I used the operating specs from the datasheet (http://www.tme.eu/en/Document/565faa25e36aa0e6e3f4693c6906e818/BZ-25SMD.pdf)

5V, 1.5mA

R = V / I = 5V / ( 1.5 / 1000 A) = 3.3K Ohm

Then using 3V3, for the new Arduinos to find current:

I = V / R = 3.3V / 3300 Ohm = 1mA
1mA < 7mA

So I think we should be okay.

One source mentioned that there's a difference between buzzers, and piezoelectric transducers. They said that piezoelectric transducers have a resistance that is ususally listed on the datasheet. The datasheet for our buzzer doesn't say it's a piezo transducer, or list a resistance. I'm going to look into this more.

aShure, 2015-05-28:
No need to, that sounds solid!

cwjpalmer, 2015-05-28:
great! thanks for the feedback :-)

closed 2015-05-28 by cwjpalmer