awesomeshield/Awesome-Shield-PCB #28: Pick New LED?

cwjpalmer, 2015-05-30:
I've been looking at parts stocks, and I'm not finding a ton of inexpensive RGB LEDs in 5050 packages that are extremely well stocked. I'm worried about having to pick a new LED in the future, due to unavailability. ie. the LED we found at TME is actually only available from TME, as far as I can tell. They have 4k in stock, but if that dries up, we need to source an alternate. Octopart page for the part:
https://octopart.com/ww-fce50tc-q1-wah+wang-51862286

My approach in looking at the other parts on the BOM has been to try to make sure we have easy, drop in replacements, from multiple suppliers, in varying quantities. This is tough with LEDs because their power requirements and footprints are so variable. Perfect drop in replacements aren't easy for me to find.

I'm wondering if we wouldn't be better of switching to a different package type that has broad and cheep availability of LEDs.

cwjpalmer, 2015-05-30:
This lite-on LED, seems like an appealing alternative. It's a very small package, though (1.6x1.6mm) https://octopart.com/search?q=LTST-C19HE1WT&view=list&start=0#_

aShure, 2015-05-30:
I found quite a lot of SMD RGB LEDs in the 5050 package at other suppliers, see here: http://www.digikey.com/product-search/en/optoelectronics/led-indication-discrete/524729?k=5050%20RGB

5050 packages are a mass produced industry standard, used a lot in LED strips.

It won't be a problem to use a different brand and model if the 5050 footprint is similar.You can have a look at other PLCC6 packages in smaller sizes. PLCC6 is the general term for those LED packages with six feet. 5050 is the biggest, usually brightest one. Another one would be 3528 or smaller. The four digits declare the XY size of the package, 5050 equals 5.0 x 5.0 mm²

The Liteon LED you linked there comes in a 1616 package. I'd prefer the 5050.

cwjpalmer, 2015-06-09:
sticking with the 5050

closed 2015-06-09 by cwjpalmer