awesomeshield/Awesome-Shield-PCB #33: Use 5V instead of IOREF

cwjpalmer, 2016-07-01:
I think it would be better to use the 5V pin to supply power to the potentiometer and the photodiode, rather than the IOREF pin.

aShure, 2016-07-06:
Why do you think it would be better?

cwjpalmer, 2016-07-08:
@aShure My thinking is that it's a good idea to keep the IOREF pin unused so that people can use it to adjust the IO reference voltage if they want. Is there an advantage to using it over the 5V to supply power to the analog components?

Maybe using the AREF pin to supply power to the analog components is actually a better way to go? That way if you change the AREF voltage you're using, you still get a full range of readings from the analog components.

aShure, 2016-07-08:
The IOREF pin can't be adjusted by the user, it is exactly for this purpose, to have a full scale reading on the ADC. The IOREF pin signals to the shield what kind of host board is used. 

Using 5V hardwired is not a good idea because host boards might run on 3V3.

aShure, 2016-07-08:
The AREF pin is meant to be used as an input, not a reference output. Different application.

cwjpalmer, 2016-07-08:
Okay great! Thanks @aShure I misunderstood, and it sounds like we should keep using the IOREF pin.

closed 2016-07-08 by cwjpalmer