awesomeshield/Awesome-Shield-PCB #35: Component & Label Spacing

cwjpalmer, 2016-07-04:
Components (excluding resistors and headers) should have enough space next to them for labels. Space for labels should be on the side closest to the end of the board that pin 0 and pin A5 are on on an Arduino Uno pinout. The photodiode resistor should be placed far enough from spaces that labels will be in to avoid ambiguity of any labels.

4 pin connectors ("Grove ports") should have enough space around them for either a label or a shape to be drawn around them on the silk screen.

cwjpalmer, 2016-07-08:
@aShure FYI the font we're going to use for component labels on the silk screen will be the summer font, which you can see in the Awesome Shield styleguide http://awesomeshield.com/styleguide (it's the H1 font). I'm pointing this out because it's pretty tall and skinny, so the aspect ratio will effect the component spacing.

cwjpalmer, 2016-07-08:
Here's a sample graphic that has the component names in the correct font.

[image]

aShure, 2016-07-24:
Needs to be done after the E-CAD, Eagle can't render fonts. It's also horrible at dithering/aliasing/interpolating anything. 
I suggest submitting Eagle sources plus rendered masks (export form Eagle, modify with graphic editor) to manufacturer.
Tipp: Use the top, bottom and guide layers as low opacity overlays when working on the silkscreen design in a graphic editor, makes it easy to spot collisions like vias.

cwjpalmer, 2016-08-01:
added bitmap top silkscreen in V0.60 release

closed 2016-08-01 by cwjpalmer