awesomeshield/Awesome-Shield-PCB #40: Router track should be continuous

cwjpalmer, 2016-07-25:
There are currently breaks in the router path to create tabs for panelization. I think that this should be added layer by the manufacturer. I think for now, we should show a continuous routing path around the edge of the board. What do you think @aShure ?

aShure, 2016-07-25:
Finished files should contain no milling layer at all (each layer is a separate file anyways, so it's ok to keep it in the source file for now if you want to mill a prototype yourself, for example in the fablab berlin). Just the dimension layer is important.
Ask the manufacturer if they build the panel or have a template at hand and if they want to have a milling layer. It's also their call what end mill to use. Liteon, Andus etc might charge extra for milling with small end mills. Usually a stack of boards is milled and drilled at once. For small end mills the stack can't be as high, so each and every panel is milled separately with a slower speed -> extra cost.

cwjpalmer, 2016-07-25:
👍  @aShure Thanks! That's really good to know.

closed 2016-07-25 by cwjpalmer